pothosware/SoapySDR #442: C# bindings for .net 8.8

VE3NEA, 2024-11-11:
I have created a Visual Studio solution to build the SoapySDR C# bindings for .net 8.0, and generated html documentation for the bindings. The solution also includes a utility in C# that enumerates the SDR devices and collects all available info about them. The repository is [here](https://github.com/VE3NEA/SoapySdrSharp). 

Please see if you can include this work in the SoapySDR project. I cannot do that because I do not have enough experience with cmake.

zuckschwerdt, 2024-11-11:
Interesting, thanks. The SoapySDR project would need to include control files (for swig it's `.i` files) and (CMake) rules, not generated files and binaries.

Can you detail which files are original (as opposed to unmodified generated files) and what process you used to generate files (swig options etc.)? What about the doc files, are those generated (by docfx) based on just the .cs files? With the C# files I'm not sure how all those are generated and if they are unmodified?

For the binary files there should be an indication where those came from, which version and how to update. (Or preferably a way to download unpack and copy over when needed as those usually would not be include with source files). 

It's nice to see this all bundled for a working example. Maybe this is more of a complete starter kit than an enhancement to SoapySDR?

VE3NEA, 2024-11-11:
There are two projects in the repo,  [Pothosware.SoapySDR](https://github.com/VE3NEA/SoapySdrSharp/tree/master/Pothosware.SoapySDR) and [ListSoapySdrDevices](https://github.com/VE3NEA/SoapySdrSharp/tree/master/ListSoapySdrDevices). For the first project I just cloned SoapySDR and ran **cmake**, then I created a new .net 8.0 Class Library project and added the existing c# files to it. The only new file in this project is [Pothosware.SoapySDR.csproj](https://github.com/VE3NEA/SoapySdrSharp/blob/master/Pothosware.SoapySDR/Pothosware.SoapySDR.csproj).

The documentation is generated by running **DocFX** in the `doc` folder, without arguments.. The doc generation settings are in the [docfx.json](https://github.com/VE3NEA/SoapySdrSharp/blob/master/Pothosware.SoapySDR/doc/docfx.json) file. It is currently configured to produce html for offline reading, but it could be changed to generate files for publishing on Github Pages.

The source code in the second project is mine. It makes use of the `Pothosware.SoapySDR.dll` library built in the first project, all other dll's were built by cloning the corresponding repositories and running **cmake** + **VS** on them. The plugin projects had to be modified to get the include and library files from the SoapySDR repository instead of those that come with the **Pothosware** deployment.

I can keep my repository for those who need the whole thing, but some parts of it, such as the .net 8.0 project and docs, could be added to the SoapySDR project.

zuckschwerdt, 2024-11-11:
Usually projects don't include files that can be readily generated. I think the most benefit for other developers trying to get started would be to have an expanded step-by-step howto, like you wrote above, along an example repo like you created.

Like e.g.
- download or clone the repo from **url**
- run `cmake ..some needes args..`
- create a project and set X to Y, add Z.
- …

VE3NEA, 2024-11-11:
On Windows we do not need to rebuild the libraries every time we want to use them. Once built, they work for everyone, on every 64-bit version of Windows. Since the dll's are already available, I do not see much value in describing the building steps in detail.

zuckschwerdt, 2024-11-11:
I'm not really a C# user and I don't develop on Windows, so I don't really know how much effort and guidance is needed to get started. It sounded like some libs would need to be collected and sorted into specific folders to get going?

VE3NEA, 2024-11-11:
Yes. To use SoapySDR in a C# app, one needs to include the bindings assembly (`Pothosware.SoapySDR.dll`) as a dependency in the project, and copy all non-C# dll's from the example app to their project. No libraries need to be recompiled.

eabase, 2025-02-14:
Hi Alex, @VE3NEA 

> On Windows we do not need to rebuild the libraries every time we want to use them. Once built, they work for everyone, on every 64-bit version of Windows. Since the dll's are already available, I do not see much value in describing the building steps in detail.

Yes, this look like a good addition, but for all sanity sake, please describe the build procedure!
I'm trying to get stuff built on Win-11 and it's huge PITA because of lacking and wrong documentation. 
I would love to know all the build steps. Also I'm building from CLI (posh) using VS Dev Environment.

VE3NEA, 2025-02-14:
To rebuild SoapSDR on Windows, run these commands in Git Bash:
```
git clone https://github.com/pothosware/SoapySDR.git
cd soapysdr
mkdir build
cd build
cmake ..
```
This will create the `SoapySDR.sln` file. Open this file in Visual studio, then in Solution Explorer right-click on the `SoapySDR` project and click on Build. Do the same with with the `Pothosware.SoapySDR` and `SoapySDRCSgaroSWUG` projects. The solution also includes some example and test projects, some of them fail to build, but you do not need them.

I have just tried these steps, and they worked fine.

eabase, 2025-02-19:
@VE3NEA

1. You last comment is confusing. Are there 3 project that need to be compiled? You say

> `SoapySDR` project 
> `Pothosware.SoapySDR` and 
> `SoapySDRCSgaroSWUG` projects

2. I'm building from MS DEV CLI, so that it can be autogenrated and scipted from a github action.
So I would need to know what exactly MS VS is doing when you jump through those clicking operations?

I was under the impression that CMake already did this, the second time you run cmake. 
We actually run cmake 3 times, like this: 

```bash
cmake -S .. -B . -Ax64 -Wno-dev -Wno-deprecated --fresh --install-prefix 'C:\Program Files\SoapySDR' -DCMAKE_BUILD_TYPE=Release -DENABLE_APPS=ON -DENABLE_CSHARP=ON -DENABLE_DOCS=OFF -DENABLE_LIBRARY=ON -DENABLE_LUAJIT=ON -DENABLE_PYTHON2=OFF -DENABLE_PYTHON3=ON -DENABLE_TESTS=OFF

cmake --build . --config Release --parallel 8
cmake --build build --config Release --target install
```

VE3NEA, 2025-02-19:
The solution had multiple projects, I just built the ones that I needed. Cannot help you with CLI and Python, sorry.