pothosware/SoapySDR #443: No device match when specifying driver argument

dwd-pete, 2024-11-29:
I am using SoapySDR through a conda environment that uses Python 3.9. If I use the latest SoapySDR package build, I have a strange error where devices are not matched/found when specifying the driver in the device arguments.

Example of the issue:
```
(helpdesk_test) dwd@airt-dev:~$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-5
API Version: v0.8.0
ABI Version: v0.8
Install root: /home/dwd/.conda/envs/helpdesk_test
Search path:  /home/dwd/.conda/envs/helpdesk_test/lib/SoapySDR/modules0.8
Module found: /home/dwd/.conda/envs/helpdesk_test/lib/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.3)
Available factories... rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
(helpdesk_test) dwd@airt-dev:~$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

(helpdesk_test) dwd@airt-dev:~$ python
Python 3.9.18 | packaged by conda-forge | (main, Dec 23 2023, 17:20:25) 
[GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import SoapySDR
>>> args = dict(driver="rtlsdr")
>>> sdr = SoapySDR.Device(args)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dwd/.conda/envs/helpdesk_test/lib/python3.9/site-packages/SoapySDR.py", line 1833, in __new__
    return cls.make(*args, **kwargs)
  File "/home/dwd/.conda/envs/helpdesk_test/lib/python3.9/site-packages/SoapySDR.py", line 1459, in make
    return _SoapySDR.Device_make(*args)
RuntimeError: SoapySDR::Device::make() no match
>>> exit()
```

Note that I can still create a SoapySDR device via a blank constructor. That is, `sdr = SoapySDR.Device()` seems to work.

The problem resolves itself completely if I downgrade the package to the previous build, as shown below.
```
(helpdesk_test) dwd@airt-dev:~$ conda install soapysdr=0.8.1=py39hd16970a_4
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/dwd/.conda/envs/helpdesk_test

  added / updated specs:
    - soapysdr==0.8.1=py39hd16970a_4

The following packages will be DOWNGRADED:

  soapysdr                             0.8.1-py39hbd2ca3f_5 --> 0.8.1-py39hd16970a_4

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
(helpdesk_test) dwd@airt-dev:~$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-4
API Version: v0.8.0
ABI Version: v0.8
Install root: /home/dwd/.conda/envs/helpdesk_test
Search path:  /home/dwd/.conda/envs/helpdesk_test/lib/SoapySDR/modules0.8
Module found: /home/dwd/.conda/envs/helpdesk_test/lib/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.3)
Available factories... rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
(helpdesk_test) dwd@airt-dev:~$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

(helpdesk_test) dwd@airt-dev:~$ python
Python 3.9.18 | packaged by conda-forge | (main, Dec 23 2023, 17:20:25) 
[GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import SoapySDR
>>> args = dict(driver="rtlsdr")
>>> sdr = SoapySDR.Device(args)
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner
>>> exit()
```

zuckschwerdt, 2024-11-29:
Do you known the git hashes the packages are built from? The information `Lib Version: v0.8.1-4` vs `Lib Version: v0.8.1-5` is packager specified (via `-DSOAPY_SDR_EXTVER`).

zuckschwerdt, 2025-02-14:
Likely a lib/modules version mismatch or packager issue. Closing for now.

closed 2025-02-14 by zuckschwerdt