pothosware/SoapySDR #446: How to build & install SoapySDR on Windows?

eabase, 2025-02-12:
Looking at the page:
https://github.com/pothosware/SoapySDR/wiki/BuildGuide

Say that:
> add `C:\Program Files\SoapySDR\bin` to the `%PATH%`

But:
`cmake --build my_build_dir --config Release --target install`

actually installs into the x86 directory:  
`C:\Program Files (x86)\SoapySDR`

**How to build SoapySDR on Windows?**

What is the optional `swig` dependency used for?
(How should that be installed?)

eabase, 2025-02-24:
Ok, managed to build the *optional* `SoapyRTLSDR`, here:
https://github.com/pothosware/SoapyRTLSDR/issues/79

Sort of works...but need more testing. (How?)

Does anyone, know how I can test this?

zuckschwerdt, 2025-02-24:
rx_sdr from https://github.com/rxseger/rx_tools should work as test tool for any SoapySDR module.

eabase, 2025-02-25:
They don't have any binary releases and the compile fails. 💩 
https://github.com/rxseger/rx_tools/issues/105

zuckschwerdt, 2025-02-25:
> Where exactly is the code that does the Python package installation?

https://github.com/pothosware/SoapySDR/blob/6e99da18c4b24129f902a5362ea2adf68abeb1e8/swig/python/CMakeLists.txt#L137-L145

zuckschwerdt, 2025-02-26:
I don't think that SoapySDR is on pypi, so no `pip`. You just get the swig module and wrapper .py installed to the Python dir.

eabase, 2025-02-26:
> You just get the swig module and wrapper .py installed to the Python dir.

That's not how it works AFAIK.  So how would you import into your pyhton code?  If it's not available by the package manager, you can't import (as you suggest in all the example code), with `import SoapySDR`. 

If it's not in pypi, normally you would do go to the project directory and do `pip install .`...

zuckschwerdt, 2025-02-26:
I didn't write any (Python) example code, sorry. I never used the Python bindings. To my understanding you can either install a package from a registry (pypi) or local source (setup.py) -- or just dump the files somewhere into Python's search path.

eabase, 2025-03-03:
I can't find this in pypi (as you said).
This doesn't make sense at all. :(
Can you find someone in this organization that could help? 
Maybe @guruofquality ?

eabase, 2025-03-06:
Hi Josh! @guruofquality 

Thank you.

I see that you are using HKLM to query paths, but I don't have any Python in that branch, my Python branch is located in HKCU. (And TBH IDK why that is. I surely installed to be available for the entire machine.)

```cmake
get_filename_component(PYTHON3_ROOT "[HKEY_LOCAL_MACHINE\\SOFTWARE\\Python\\PythonCore\\${PYTHON3_VERSION}\\InstallPath]" ABSOLUTE)
...
set(PYTHON3_INCLUDE_DIR ${PYTHON3_ROOT}/include)
set(PYTHON3_LIBRARY ${PYTHON3_ROOT}/libs/python${PYTHON3_VERSION_NO_DOT}.lib)
set(PYTHON3_INSTALL_DIR lib/python${PYTHON3_VERSION}/site-packages)
```

So a few things here, right away.

- You are mixing Windows and Posix paths. `\` vs `/`, this never ends well..
- The name is also outdated with the **`3`**. No sane person is still using Python2, so let's avoid using that notation, as it is no longer portable or compatible. 
- Check other windows installs for what registry entries they are using. 
- Consider **not** using Registry entries at all. Having you Pythin in the path or setting the Python specific environment variables should be enough.
- Why is `PYTHON_INSTALL_DIR` not including the `PYTHON_ROOT`?

---

So this means that I should have:

```bash
PYTHON_ROOT='C:\lang\Python312'
PYTHON_INCLUDE_DIR='C:\lang\Python312\include'          # These are private Python distro includes.
PYTHON_LIBRARY='C:\lang\Python312\libs\python312.lib'   # Not the "python3.lib" or "_tkinter.lib"
PYTHON_INSTALL_DIR='C:\lang\Python312\site-packages'    # Where pip packages go...
```

Then we have:
- in `.\Lib\<packagename>`: **private** to Python built-ins.
- in `.\Lib\site-packages\<packagename>`: we have the `__main__.py` and `__init__.py` (package) files

For simplicity let's call the files: 
**`A`** = `__init__.py` - Contains a *Class* with the *importable* name.
**`B`** = `__main__.py` - Contains the *main* entry point. 

So AFAIK, in order to be able to use a statement like `import SoapySDR`, we have to:
1. Place the 2 files `A` and `B` into the `<packagename>` sub-directory of `.\site-packages\`.
2. File `A` Contains a *Class* with the *importable* name:

```python
class SoapySDR:
    def __init__(self, config_file=None):
        config_file = Path(config_file) if config_file is not None else None
        self.config = Configuration(config_file)
    ...
```

3. File `B` contains the import and then *main* code entry point:

```python
from packagename import SoapySDR
...
if __name__ == "__main__":
    print("The Main code starts here...")

```

It doesn't seem we have any such Class, or function...
So how (and where) did you generate and install these files so that they can be imported with `import SoapySDR`?