pothosware/SoapySDR #448: [documentation] How to build and install python package in native Windows-11 x64?

eabase, 2025-02-14:
I've tried to build the whole package using VS on Windows-11 via PowerShell VS DEV shell. 
However, the Python packages are not built installed and there are several other issues.
Also, I'm also trying to build using the `LimeSDR`, and [first issue](https://github.com/myriadrf/LimeSuiteNG/issues/118) was posted there, and I was told most issues are from you guys.  

For example, the SoapySDR package is installed in the wrong location, in:  
`C:/Program Files (x86)/SoapySDR` instead of **`C:/Program Files/SoapySDR`**. 

How can I get a `SoapySDR` - or Python compatible Python API with this device? 

Or is it `PothosSDR`?  What's the difference??

* https://github.com/pothosware/SoapySDR
* https://github.com/pothosware/SoapySDR/wiki/BuildGuide

When running install (Windows powershell), I get this:

```bash
# cmake -B mybuild

-- Building for: Visual Studio 17 2022
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.26100.
-- The C compiler identification is MSVC 19.42.34435.0
-- The CXX compiler identification is MSVC 19.42.34435.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.42.34433/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.42.34433/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
--
-- #############################################
-- ## Begin configuration for C# support...
-- #############################################
-- Enabling optional C# bindings if possible...
-- Looking for a CSharp compiler
-- Looking for a CSharp compiler - C:/Program Files/Microsoft Visual Studio/2022/Community/MSBuild/Current/Bin/Roslyn/csc.exe
-- Could NOT find SWIG (missing: SWIG_EXECUTABLE SWIG_DIR)
-- SWIG_FOUND: FALSE -
CMake Deprecation Warning at swig/python/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Could NOT find SWIG (missing: SWIG_EXECUTABLE SWIG_DIR)
-- SWIG_FOUND: FALSE -
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
--
-- #############################################
-- ## Begin configuration for Python 2 support...
-- #############################################
-- Enabling optional Python 2 bindings if possible...
-- Could NOT find Python2 (missing: Python2_EXECUTABLE Python2_LIBRARIES Python2_INCLUDE_DIRS Interpreter Development)
--
-- #############################################
-- ## Begin configuration for Python 3 support...
-- #############################################
-- Enabling optional Python 3 bindings if possible...
-- Found Python3: C:/lang/Python312/python.exe (found version "3.12.9") found components: Interpreter Development
--  * Interpreter: C:/lang/Python312/python.exe (Python)
--  * Include: C:/lang/Python312/include
--  * Library: C:/lang/Python312/libs/python312.lib
--
-- #############################################
-- ## Begin configuration for LuaJIT support...
-- #############################################
-- Enabling optional LuaJIT bindings if possible...
CMake Deprecation Warning at luajit/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Could NOT find LuaJIT (missing: LUAJIT_INTERPRETER)
--
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
--
 * Library, runtime library v0.8.1-g309335ec
 * Apps, command line applications
 * Tests, library unit tests

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
--
 * Docs, doxygen documentation
 * CSharp, C# bindings v4.7.2                                    <----- WHY IS THIS DISABLED?
 * Python2, Python2 bindings v
 * Python3, Python3 bindings v3.12.9                             <----- WHY IS THIS DISABLED?
 * LuaJIT, LuaJIT bindings

-- SoapySDR version: v0.8.1-g309335ec
-- ABI/so version: v0.8-3
-- Install prefix: C:/Program Files (x86)/SoapySDR               <----- WRONG Location for x64
-- Configuring done (12.7s)
-- Generating done (0.2s)
-- Build files have been written to: C:/xxx/xxx/SoapySDR/mybuild

```
It seem that your documentation for building on Windows need to be updated. Nothing can be built and used as they are currently presented. There are essential dependencies missing.

**How can I correctly build, install and use the Python package and bindings?**

---
<sub> PS. I use `LimeSDR USB v1.4s` and and `Nooelec RTLSDR USB v4`.</sub>

eabase, 2025-02-23:
@zuckschwerdt 
Do you know if I need to build the other libraries (such as `rtlsdr.dll` and `LimeSuiteNG.dll`) **first** (before SoapySDR)?

zuckschwerdt, 2025-02-23:
The libraries you listed are only needed to *run* modules. You need the matching headers to *build* modules, but not a dependency for SoapySDR itself.

eabase, 2025-02-24:
Well, I'm trying to get the python import to work, so I have no idea what I need to build with...