pothosware/SoapySDR #451: [windows] Doxyfiles contain deprecated items causing "warning 560: Unknown Doxygen command"

eabase, 2025-02-19:
Doxyfile contains several items that doxygen claims to be unknown. 

I can only assume they are very old and have been deprecatred long ago. 
(Please correct me if I'm worng.)

To fix, run: 
```bash
# After 1st cmake
doxygen -u .\swig\csharp\docs\Doxyfile
```
The input files are: 

```bash
# In files:
./docs/Doxyfile.in
./swig/csharp/docs/Doxyfile.in
```

eabase, 2025-02-20:
@zuckschwerdt 
Awesome! TY.

eabase, 2025-02-23:
There's still an issue here. You say: 

> which has been replaced by TIMESTAMP in release 1.9.7 (release date 18-05-2023) but is still honored (with 1.11.1).

But latest doxygen is ~ `1.13.2`, so I still get: 

```bash
CUSTOMBUILD : warning : Tag 'HTML_TIMESTAMP' at line 1150 of file 'C:/xxxx/SoapySDR/build/docs/Doxyfile' has become obsolete. [C:\xxxx\SoapySDR\build\docs\docs.vcxproj]
```

In addition, your PR didn't address similar issues in the file: 
`..\swig\csharp\docs\Doxyfile.in`. 

https://github.com/pothosware/SoapySDR/blob/master/swig/csharp/docs/Doxyfile.in

---

Then we have the `warning 560: Unknown Doxygen command` for all the  `parblock` and `endparblock` entries in the 2 files: 

```bash
./include/SoapySDR/Device.h    # not used? 
./include/SoapySDR/Device.hpp  # warning 560
```

See warnings here:

```bash
SoapySDRCSHARP_wrap.cxx
C:\xxxx\SoapySDR\include\SoapySDR\Device.hpp(234): warning 560: Unknown Doxygen command: parblock. [C:\xxxx\SoapySDR\build\swig\python\python3\_SoapySDR3.vcxproj]
C:\xxxx\SoapySDR\include\SoapySDR\Device.hpp(252): warning 560: Unknown Doxygen command: endparblock. [C:\xxxx\SoapySDR\build\swig\python\python3\_SoapySDR3.vcxproj]
C:\xxxx\SoapySDR\include\SoapySDR\Device.hpp(255): warning 560: Unknown Doxygen command: parblock. [C:\xxxx\SoapySDR\build\swig\python\python3\_SoapySDR3.vcxproj]
C:\xxxx\SoapySDR\include\SoapySDR\Device.hpp(259): warning 560: Unknown Doxygen command: endparblock. [C:\xxxx\SoapySDR\build\swig\python\python3\_SoapySDR3.vcxproj]
C:\xxxx\SoapySDR\include\SoapySDR\Device.hpp(261): warning 560: Unknown Doxygen command: parblock. [C:\xxxx\SoapySDR\build\swig\python\python3\_SoapySDR3.vcxproj]
C:\xxxx\SoapySDR\include\SoapySDR\Device.hpp(265): warning 560: Unknown Doxygen command: endparblock. [C:\mydev\gitclones\SoapySDR\build\swig\python\python3\_SoapySDR3.vcxproj]
```

zuckschwerdt, 2025-02-23:
The warning about `HTML_TIMESTAMP` is expect and will stay. It's only one single warning.

The `\parblock` seem broken in the headers, I'll fix those.

zuckschwerdt, 2025-02-23:
Actually the `warning 560: Unknown Doxygen command: parblock` is a Swig warning and must be addressed in the Swig project.

closed 2025-02-23 by zuckschwerdt

eabase, 2025-02-24:
Why would Swig warn about Doxygen stuff?
What is the problem exactly?

I'm happy to file report at Swig, if you point me the right direction...

zuckschwerdt, 2025-02-24:
Swig has a Doxygen parser, search "warning 560: Unknown Doxygen command"

eabase, 2025-02-24:
Indeed:

https://www.doxygen.nl/manual/commands.html#cmdparblock

Filed issue:
https://github.com/swig/swig/issues/3130

PR #452 (2025-02-23): Update Doxygen config to remove obsolete settings
Removes Doxygen settings that are obsolete and were default anyway to reduce clutter.

Drops
- `TCL_SUBST`
- `CLANG_ASSISTED_PARSING`
- `CLANG_OPTIONS`
- `COLS_IN_ALPHA_INDEX`
- `FORMULA_TRANSPARENT`
- `LATEX_SOURCE_CODE`
- `LATEX_TIMESTAMP`
- `RTF_SOURCE_CODE`
- `DOCBOOK_PROGRAMLISTING`
- `PERL_PATH`
- `CLASS_DIAGRAMS`
- `MSCGEN_PATH`
- `DOT_FONTNAME`
- `DOT_FONTSIZE`
- `DOT_TRANSPARENT`

Does not drop

- `HTML_TIMESTAMP`

which has been replaced by `TIMESTAMP` in release 1.9.7 (release date 18-05-2023)  but is still honored (with 1.11.1).

Cl [... cut]