pothosware/SoapySDR #457: [Ubuntu] Compatibility with RX-888 MKII

AlexandroRabadanParra, 2025-03-10:
I'm really new to working with SDRs and even Linux. Is the RX-888 MKII compatible with SoapySDR?

My goal is to use the API with Python on an Ubuntu machine. I have installed SoapySDR with Anaconda, but after trying some things, I am not able to make SoapySDR find my device. What are the steps to make it work?

zuckschwerdt, 2025-03-10:
It should be. You need to install a driver module for the hardware. https://github.com/cozycactus/SoapyRX888

closed 2025-03-10 by zuckschwerdt

fventuri, 2025-03-10:
There's also SoapySDDC in the main repository for the RX888 firmware (https://github.com/ik1xpv/ExtIO_sddc/tree/master/SoapySDDC), which is based on the code developed by @cozycactus
He also reported that it works for him on Ubuntu 22.04: https://github.com/ik1xpv/ExtIO_sddc/pull/234#issuecomment-2224242725

Franco