pothosware/SoapySDR #459: Low-level library

AhmedSoliman3, 2025-03-18:
I see in Python Soapysdr can access low-level libraries, while in CPP I can't access Low-level libraries, I can't control registers like "INPUT_CTL_PGA_RBB", "CCOMP_TIA_RFE" and others. I hope someone can help me

zuckschwerdt, 2025-03-18:
You might be looking for the Register API, LimeSDR has this implemented: https://github.com/myriadrf/LimeSuite/blob/master/SoapyLMS7/Settings.cpp#L760-L811

AhmedSoliman3, 2025-03-20:
thanks, i found the issue was what is the name of an available register interface to use and how to use them, i found listRegisterInterfaces function returns the registers name, "RFIC0 , BB". then i used readRegister("RFIC0", address); to read register value. also writeRegister("RFIC0", address, regValue);

closed 2025-03-20 by AhmedSoliman3