pothosware/SoapySDR #461: SoapySDR::Device::writeStream  is the numElements based on the data type choosen?

smileyrekiere, 2025-04-11:
When working with SoapySDR with hackrf and BladeRF.

1. for hackrf using SOAPY_SDR_CS8 for creating I/Q values.  Thus the number of bytes equals the number of elements passed on write stream. 
2. for Bladerf using SOAPY_SDR_CS16 (soapySDR does not support the 8 bit mode for the BladeRF).  Each element of the I/Q value is 16 bits , so would the number of element be the number of I/Q values or the number of elements be the number of bytes?

Note for SOAPY_SDR_CS16 the SoapyBladeRF will convert  it to BLADERF_FORMAT_CS16_Q11_META in bladeRF_Streaming.cpp , I hacked the bladeRF_Streaming.cpp to use BLADERF_FORMAT_CS16_Q11 , since have not figured out how to pass the meta mode (auto or normal). 

3. for the buffer size on the hackrf for writeStream pass in the number of bytes for the number of elements and the hackrf performs as expected (the fixed buffer size of 262144) (256K alignament)

the bladeRF needs to have a buffer_size_min = (2 x num_samples * num_channels * sizeof(int16t))  , only using one channel so 2048 samples  , 2 * 2048 * 1 * 2 = 8192 .  If there are 2048 I/Q pairs would the number of elements be 2048, 4096 or 8192.

zuckschwerdt, 2025-04-11:
Elements, `numElems` in the code, are the most basic and indivisible unit of SDR data. They are of size `(I+Q)`, i.e. `2 * (sizeof data_type)`. E.g. 16 bit for CS8, 32 bit for CS16, 24 bit for CS12 (note esp. that last one).

In hardware terms a channel might well be just a single ADC, i.e. 2 channels form an I/Q signal, but in SoapySDR terms channels refers to multiple RX or TX channels (each of I/Q).

closed 2025-04-15 by zuckschwerdt

smileyrekiere, 2025-04-20:
Thank-you for the response it was insightful,  in particular the comment about the ADC and the measure of the elements.
o.k. for the hackrf  the nubmer of elements for the buffer is 131072 which has an MTU of 262144.   Hackrf is fixed at the element size of 131072. 

So for single channel with the BladeRF each element is 4 bytes.  Does the SoapyBladeRF have a fixed MTU / default MTU size.  The pulses look funny.  Does Blade RF have a fixed MTU size ?

Hackrf pulses 10 us and leading edge 100 us appart. six pulses 

[image]

BladeRF pulses 10 us and leading edge 100 us apart six pulses , yet doesn't look that way , varying pulse widths and separation of pulses.  

[image]

Any insight helpful