pothosware/SoapySDR #462: Soapy

smileyrekiere, 2025-04-21:
SoapySDR writeStream pulses are not as expected when using the Blade RF

from Issue: 
https://github.com/pothosware/SoapySDR/issues/461
The response provided by zuckschwerdt

`Elements, numElems in the code, are the most basic and indivisible unit of SDR data. They are of size (I+Q), i.e. 2 * (sizeof data_type). E.g. 16 bit for CS8, 32 bit for CS16, 24 bit for CS12 (note esp. that last one).

In hardware terms a channel might well be just a single ADC, i.e. 2 channels form an I/Q signal, but in SoapySDR terms channels refers to multiple RX or TX channels (each of I/Q).`

So each CS16 element is 4 bytes which is the unit used by the BladeRF for SOAPY_SDR_CS16.

The response it was insightful, in particular the comment about the ADC and the measure of the elements.

Based on the above information 
For the hackrf each element is 2 bytes the number of elements for the buffer is 131072 which has an MTU of 262144 bytes. Hackrf is fixed at the element size of 131072 which needs to be padded out to that value. 

So for single channel with the BladeRF each element is 4 bytes. **Does the SoapyBladeRF have a fixed MTU / default MTU size?**  I did not think , tried many variations.

Thus

Hackrf pulses 10 us and leading edge 100 us appart. six pulses

[image]

For the Blade RF tried multiple buffer/MTU sizes with padding and one element being 4 bytes

[image]

Did look at bladeRF_SoapySDR::setupStream since creating the I/Q values I forced BLADERF_FORMAT_SC16_Q11 since BLADERF_FORMAT_CS16_Q11_META Since I'm not sure if the I/Q values are being replaced based on the following comment for META type data.

`This format is the same as the ::BLADERF_FORMAT_SC16_Q11 format, except the first 4 samples in every <i>block*</i> of samples are replaced with metadata organized as follows. All fields are little-endian byte order.`

Bottom line is the I/Q values in the buffers is not being correctly to the BladeRF and trying to figure out why?  Trying to get some foot hold onto where to look?  

followed this on the creation of various buffer sized for the BladeRF

https://www.nuand.com/libbladeRF-doc/v2.5.0/group___s_t_r_e_a_m_i_n_g___f_o_r_m_a_t.html

smileyrekiere, 2025-04-24:
Thank-you, ,   the BladeRF Soapy looks to default to 32 buffers, So is it necessary to have data for all 32 buffers?

zuckschwerdt, 2025-04-24:
That's just rotating transfer buffers, nothing to do with the Soapy API.

closed 2025-04-24 by smileyrekiere