pothosware/SoapySDR #466: Is there any updated Icom modules available?

ArloG3300, 2025-06-19:
#421 from 2023 asks if there are any updated Icom IC-R8600 drivers available.
I'm only seeing ones from 7 years ago for x64 & x86 Windows.
Several newer SDR applications are using 0.8.x modules.
Is there a method with tutorials available to create updated .dll files for the newer modules?

And if so (hoping very much so) can other Icom radios get newer drivers?

ArloG3300, 2025-07-02:
Please. Can anybody help?
The developer of SkyRoof suggested that I ask here.
He reworked a 0.7 module to 0.8 but the radio does not operate using the I/Q port.
Also SDR++ Brown is not working with the available module and SoapySDR support file.
The radio shows that it is connected to the program. But no response in the program.
When the program is closed (Windows 11). The I/Q port does not release.

If this is something I need to try myself. Is there a clear step-by-step tutorial that may assist me in learning?

zuckschwerdt, 2025-07-02:
Compiling the SoapySDR module on 0.8 should still be possible. Compiling with Visual Studio and CMake should just work.

On other platforms you might need two fixes though:
```diff
--- a/Settings.cpp
+++ b/Settings.cpp
@@ -144,7 +144,7 @@ void SoapyICR8600::setAntenna(const int direction, const size_t channel, const s
        }
        else {
                if (name != "ANT 1") {
-                       SoapySDR_logf(SOAPY_SDR_ERROR, "setAntenna %s invalid for frequency %d", name, centerFrequency);
+                       SoapySDR_logf(SOAPY_SDR_ERROR, "setAntenna %s invalid for frequency %d", name.c_str(), centerFrequency);
                }
        }

--- a/SoapyICR8600.hpp
+++ b/SoapyICR8600.hpp
@@ -89,7 +89,7 @@ public:
                long long &timeNs,
                const long timeoutUs = 100000);

-       int SoapyICR8600::readData(SoapySDR::Stream* stream, void* buff);
+       int readData(SoapySDR::Stream* stream, void* buff);

        /*******************************************************************
```

zuckschwerdt, 2025-07-04:
The Icom IC-R8600 seems to be rare and expensive, that may be why you can't find people interested in it.

Did you try to use a cheap ubiquitous SDR like the RTL-SDR to get acquainted with the SDR ecosystem?

I haven't used Visual Studio and I don't have a Windows system available to develop things. But I was successful in just compiling the code from https://github.com/righthalfplane/SoapyICR8600 -- just two lines need changes as the GCC compiler is stricter than Visual Studio.
You need the SoapySDR library code, either compile yourself or get the DLLs available from [Pothos downloads](https://github.com/pothosware/PothosCore/wiki/Downloads).
Then compile SoapyICR8600 with that source. It will be bound to that version then.
There's no guarantee that the module then works -- as you said e.g. the frequency controls might be broken or such.

ArloG3300, 2025-07-05:
Pardon me. I do not wish to veer off topic. Expensive yes. "Rare" not entirely. A very nice radio with excellent characteristics and sensitivity.Interest in them and other higher end radios that use the I/Q interface. Same as any SDR radio. Is avid, but a niche.
Although not exactly ubiquitous nor inexpensive. I owed my SDRPlay RSP2 for some time before purchasing the R8600 to supplement the very nice IC-R8500 that I enjoyed for many years and performs very nice after a go-over by a former and enthusiastic Icom tech.
So yes. I am Very acquainted in SDR software and use many different programs such as HDSDR, SDR-Console, Older versions of SDR# that supported the R8600. SDR++. Many actually do support the IC-R8600.

I have never used Visual Studio. And information of exactly what is needed and the specific tutorials to get a working software set needed to get my own radio working. Such as is the case with anything SoapySDR are just nonexistent. Or worded for the experienced.

You mention you were successful in compiling the code, you changed just two lines in a GCC compiler.
SoapySDR Library code? Compile myself? I have the entire SoapyICR8600-master library that includes .py, .md, hpp, cpp....files. I have 0 clue currently of the most remote idea of how to use them. And in the directory are included 32 & 64 bit icr8600Support.dll files.
The question is. Were the included icr8600Support.dll files the results of using the files from the SoapyICR8600-master library? If not then where did they originally come from.

What I can say is that I used a decompiler to compare RTL based .dll files with the factory IC-R8600 .dll files.
I understand serial communications. I've sniffed the USB I/Q port when using SDR programs.
I do not understand how I/Q data over USB turns into audio in software or how to troubleshoot when it does not.

"Frequency controls not working". My big wonder. An example is if the SDR software sent a hex code out to an RTL dongle to up the frequency by x kHz. And the radio responds with another string to tell the software it has.To keep from repeating the command packet.
Lines in the .dll show the hex. As they do in the Icom .dll.
Yet one works and the other will not. Unless it's "Compiled with...."

You are correct. Interest may be limited. But HDSDR always works. SDR-Console always works. WFview using the CI-V protocol always works. And many programs using SoapySDR don't. With drivers not updates since 2018.

What are the building blocks for a person like me to get me personally on the right track to build and debug the modules and bits and pieces to get these applications working. I'm sure versions from 2018 worked well. And in 2025....they don't.

Make any sense at all?

zuckschwerdt, 2025-07-06:
The `icr8600Support.dll` is what you expect the SoapyICR8600 module to compile to. It needs to match the SoapySDR version installed and must be installed or placed in the SoapySDR modules directory.

You should only need to build SoapySDR or install a Pothos package. Then build the SoapyICR8600 module. There should be guides on how to build with CMake in Visual Studio.

There are some nice up to date Windows building instructions for rtl_433 in https://github.com/merbanan/rtl_433/issues/3215 -- maybe you glean something from that. And linked are older instructions for a SoapySDR module: https://github.com/pothosware/SoapyRTLSDR/issues/79

zuckschwerdt, 2025-07-12:
I don't see and "prejudice" against higher end radios. When dedicated hackers have access to hardware you might get some interesting software. Not many hackers have access to expensive or niche hardware thus you'll see less ideas, software, and support. Expect that if some hardware is cheap and available that a lot of people will tinker with it.

Hardware might have some documented API, or people dig in and discover things. S.a. https://www.rtl-sdr.com/about-rtl-sdr/
One might research if the IC-R8600 has API documentation. Then research the SoapySDR API. Then bring them together.
The benefit would be an adapter to all SoapySDR-enabled software. That's likely where the journey for this person ends. Problem solved. There is likely no incentive to go further. For most then documentation, support, ready-made builds are not part of solving some an interesting puzzle.

As there is no immediate to-do for the core project I'm closing the issue. But feel free to further document your progress or ask questions. You might want to ask Sky Fox Icom about I/Q port and Extio, the SoapyICR8600 author about state and compatibility, and SDR++ community about source selection.

closed 2025-07-12 by zuckschwerdt