pothosware/SoapySDR #467: Issues with RX-888 MK II

AlexandroRabadanParra, 2025-06-23:
I'm trying to prepare a Python script that takes some samples with an RX-888 MK II and then perform an FFT to obtain the frequency components of the captured signal. 

To do this, I have used cozycactus's librx888, SoapyRX888 and rx888_test to load the firmware he provides into the SDR.

I'm feeding the HF input with a tone of 50 MHz to check whether it's working. But when I perform the FFT, I obtain the same result regardless of whether the tone generator is connected to the input or not: a flat spectrum without any trace of the 50 MHz peak. It seems as if the SDR is retrieving only noise data from the ADC. Like it's not able to obtain data from any of the inputs.

[image]

Checking the signal captured by the SDR, I've noticed that its only retrieving real samples with zero imaginary component.

Also, I've tried to call "sdr.setFrequency(SOAPY_SDR_RX, 0, 50e6)" and then "sdr.getFrequency(SOAPY_SDR_RX, 0)". But the frequency returned is equal to 0 Hz, something strange.

When executing SoapySDRUtil --probe="driver=rx888" in Bash, I obtained the following:

[image]

If you notice, in the lowest part of the output, it says "Full freq range:  MHz". It's not retrieving any frequency range.

Do you have any idea of what could be happening with this? Making this work is part of my master's thesis, so any help is really appreciated.

If you need to take a look at my code, it is basically the example for python in this repository slightly modified:

[image]

AlexandroRabadanParra, 2025-06-24:
Thank you very much, both of you. Applying the suggested patch solved the problem. Now I'm able to obtain the input tone on the FFT.

The RX-888 MK II is a key part of my master's thesis. Previous students made some developments for the HackRF and SDRPlay, but for our application (detection of solar radio bursts) , they turned out to be too noisy. The RX-888 MK II seems promising in that aspect.

My band of interest is under 65 MHz, so using the HF input is perfect for me. If it shows good results in that band, maybe in a future it could be considered to expand its range looking for a way to use the VHF. 

It's really likely that later in my work I'll have more doubts, so I'm sure that i will ask some questions in 'NextGenSDRs'. @fventuri, thank you for sharing it.

fventuri, 2025-06-24:
@AlexandroRabadanParra - happy to hear that you were able to solve the problem!

There are at least a couple of people in that group using their RX888's for radioastronomy and detection of solar bursts.

One of them (Nathan Towne) wrote a program called 'rx2fits' (https://sourceforge.net/projects/rx2fits/) that can be used to monitor solar bursts (see here for instance: https://groups.io/g/NextGenSDRs/message/2107).

Franco

AlexandroRabadanParra, 2025-06-25:
Thank you @fventuri. Some time ago, I looked inside the code of rx2fits to check if I could adapt it to the specs of an analog receiver that I want to emulate. But it's written in C# and Visual Basic .NET, two languages that I'm not used to. But now that my Python code is working, I can stick with it for now.

The thing is, now, with the Python code, I'm getting some error codes telling me that some overflows are being produced in the internal buffer. I just need to retrieve from the SDR 1024 samples every 0.25s, which my Python code is able to do. The problem comes from the fact that despite that, the internal buffer keeps getting filled and Python is not fast enough to take the unused samples and discard them. Therefore I'm getting overflow error codes.

The only workaround that I've found id deactivating and reactivating the streaming of data to clean the buffer every 0.25s. But it consumes some computing time that I use to retrieve more samples in each iteration.

Is there any way to clear the internal buffer without deactivating and activating the streaming?