pothosware/SoapySDR #469: Add new release tag

KarlL2, 2025-09-02:
SoapyBladeRF was updated to 0.4.2 and is available on Debian 13, but SoapySDR is still 0.8.1, meaning that `SOAPY_SDR_USER_FLAG0` & `SOAPY_SDR_USER_FLAG1` are not defined, and as a result `BLADERF_META_FLAG_RX_HW_MINIEXP1` & `BLADERF_META_FLAG_RX_HW_MINIEXP2` are not surfaced:

```
#if defined(SOAPY_SDR_USER_FLAG0) and defined(SOAPY_SDR_USER_FLAG1)
if ((md.status & BLADERF_META_FLAG_RX_HW_MINIEXP1) != 0) flags |= SOAPY_SDR_USER_FLAG0;
if ((md.status & BLADERF_META_FLAG_RX_HW_MINIEXP2) != 0) flags |= SOAPY_SDR_USER_FLAG1;
#endif
```
https://github.com/pothosware/SoapyBladeRF/blob/3d36e4209e32f3823573e6cdc2093575ec398d44/bladeRF_Streaming.cpp#L395

We would need a new SoapySDR version, that contains those new flags (added in https://github.com/pothosware/SoapySDR/commit/e622fb8277d043b552b223f067d2c9c39b3b9599), and once that updated package is available, the SoapyBladeRF package will need to be recompiled (meaning probably publish a new version?).

zuckschwerdt, 2025-09-02:
@guruofquality are you available to prepare new versions for these (and SoapyRTLSDR), or should I take this?

zuckschwerdt, 2026-10-10:
SoapySDR and all module CMake, Github Actions, and Debian builds are updated.

The SoapySDR 0.9.0 version is tagged now. Also in all modules got a minor bump for updated CMake support.

- SoapySDR  `0.8.1` -> `0.9.0`
- SoapyAirspy `0.2.0` -> `0.2.1`
- SoapyAirspyHF `0.2.0` -> `0.2.1`
- SoapyAudio `0.1.1` -> `0.1.2`
- SoapyBladeRF `0.4.2` -> `0.4.3`
- SoapyFCDPP `0.1.1` -> `0.1.2`
- SoapyHackRF `0.3.4` -> `0.3.5`
- SoapyNetSDR `0.2.0` -> `0.2.1`
- SoapyOsmo `0.2.5` -> `0.2.6`
- SoapyPlutoSDR -> `0.2.2` -> `0.2.3`
- SoapyRTLSDR `0.3.3` -> `0.3.4`
- SoapyRTL`TCP 0.1.0` -> `0.1.1`
- SoapyRedPitaya `0.1.1` -> `0.1.2`
- SoapyRemote `0.5.2` -> `0.5.3`
- SoapyUHD `0.4.1` -> `0.4.2`

closed 2026-10-10 by zuckschwerdt

KarlL2, 2026-10-12:
@zuckschwerdt Thanks. How about LimeSDR? I suppose https://github.com/myriadrf/LimeSuite/tree/master/SoapyLMS7 and https://github.com/myriadrf/LimeSuiteNG/tree/develop/plugins/soapysdr will have to be updated too?

zuckschwerdt, 2026-10-12:
Soapy modules should run without any changes. I'll try to PR a set of updates (build system) to the modules projects in the next days.